robotology/assistive-rehab #169: Extend the lib to include pixels information

pattacini, 2019-02-28:
From #166 it turns out that it might be beneficial for gaze tracking purposes to extend the lib and include pixels information along with 3D coordinates in the camera frame.

This way, the transformation of the target 3D point from the camera to the root frame - which relies on the reading from the gaze state port - is no longer needed since we can then use pixels as inputs.

As a result, we should be able to increase control bandwidth.

Once the lib will be extended, we need to update `attentionManager` and `test-tracker`.

cc @vvasco @randaz81 @valegagge

pattacini, 2019-03-04:
We experienced quite an improvement once the pixel is used for gaze tracking.

closed 2019-03-04 by pattacini

pattacini, 2019-03-04:
@valegagge the PR's been merged, hence you could think of relying on the pixel information to issue tracking.

Just a couple of references as to do it:

https://github.com/robotology/assistive-rehab/blob/f72f03d9ef2a89fa955a6bd823f0ccc19d50c943/tests/test-tracker.cpp#L85

https://github.com/robotology/assistive-rehab/blob/f72f03d9ef2a89fa955a6bd823f0ccc19d50c943/tests/test-tracker.cpp#L106-L123

valegagge, 2019-03-04:
@pataccini thanks very much