robotology/assistive-rehab #177: Assess the impact of a walking stick onto OpenPose

pattacini, 2019-03-29:
Accurately verify whether a person walking with the aid of a stick could impair somehow OpenPose in detecting body parts.

vvasco, 2019-04-03:
OpenPose is very robust to walking sticks! The following figures show the output of `yarpOpenPose` in several scenarios, of increasing difficulty.

### Walking naturally with a walking stick
The skeleton is very robustly detected and never confused with the stick:
[image]

### Stick very close to the leg and partially occluding it
The skeleton is impressively robust and the right leg's keypoints get never confused with the stick!
[image]

### One foot not visible
In this scenario, the right foot is not visible and the walking stick might be easily detected as part of the leg.
We can see that the ankle keypoint is detected even if not visible. However this is not dependent on the stick, as the ankle keypoint is still detected even when the stick is moved far from the leg. 
[image]

I think that OpenPose is very robust for the walking tests considered in our scenarios ([TUG](#173), 10MWT, 6MWT). The last scenario was reported to evaluate the potential of OpenPose in very challenging situations, but is unlikely to occur in the mentioned walking tests.

pattacini, 2019-04-03:
Nicely done!

closed 2019-04-03 by pattacini