robotology/assistive-rehab #178: Compute and validate the step length

vvasco, 2019-04-09:
A metric to extract for evaluating the quality of the movement is the step length, defined as the distance between two consecutive heel strikes. 
We want to compute the step length and validate it. The validation experiment considers a user walking at a known step length, which can be then compared with the computed one.

vvasco, 2019-04-12:
The following three experiments were performed, with a person walking at different step lengths:
1. step length fixed at ~0.5 m;
2. step length fixed at ~0.25 m;
3. step length progressively decreasing from ~0.5 m to ~0.3 m to ~0.2 m. 

The step length was computed as distance between right and left heels, using the 3D information (#183).

#### Experiment 1
Since the step length is defined as the distance between right and left heels when they strike the floor, we computed the distance instantaneously and extracted the maximum values. 
The following plot shows the instantaneous length (blue), its filtered version (red), filtered using an anti-causal filter, and the maximum values which provide the estimates of the step length.
We get a mean value of 0.49 m, which is very close to the real one (~0.5 m).

[image]

#### Experiment 2
Following the same approach, we get a mean value of 0.29 m, which is very close to the real one (~0.25 m).
[image]

#### Experiment 3
The plot shows the expected trend, with values decreasing from 0.57 m (vs. ~0.5 m) to 0.39 m (vs. ~0.3 m) to 0.19 m (vs. ~0.2 m).
[image]

Overall, the step length estimation seems quite good, also validating #183.

pattacini, 2019-04-12:
Very good results that make us hope exploiting feet in the skeleton will be doable in our context!

closed 2019-04-12 by pattacini

pattacini, 2019-04-14:
It may be beneficial to drop in here an archive containing the scripts used for this analysis.

vvasco, 2019-04-15:
Here it is:
[compute_steplength.zip](https://github.com/robotology/assistive-rehab/files/3078965/compute_steplength.zip)