robotology/assistive-rehab #179: Evaluate the foot keypoint estimation provided by openPose

vvasco, 2019-04-09:
Computing the range of motion of the ankle requires the estimation of the foot keypoint. The version of openPose we are currently using does not include feet estimation. 
Foot estimation is now available in openPose and we want to evaluate it in order to include it in the motion analysis.

vvasco, 2019-04-11:
Foot keypoints provided by openPose are:
- `BigToe`, `SmallToe` and `Heel`

The following figure shows the output of openPose for a person walking towards right. 
Segments connecting `Ankle-Heel` and `Ankle-BigToe` are highlighted for left and right foot. 
When one foot is occluded by the other, keypoints are missed (this occurs for the left foot), but we can cope with this within our framework.
For validating this, I computed the range of motion for left and right foot between the vector formed by ankle and big toe and an horizontal reference direction centered on the ankle.

[image]

The range of motion has the expected trend, coherent with what reported in [literature](https://musculoskeletalkey.com/biomechanics-2/).

closed 2019-04-12 by vvasco

pattacini, 2019-04-14:
It may be beneficial to drop in here an archive containing the scripts used for this analysis.

vvasco, 2019-04-15:
Here it is:
[visualize_2d_foot.zip](https://github.com/robotology/assistive-rehab/files/3078982/visualize_2d_foot.zip)