robotology/assistive-rehab #183: Evaluate the 3D projection of the foot keypoint

vvasco, 2019-04-11:
Given the foot estimation in 2D seems accurate (#179), we want to accurately evaluate if the 3D projection impairs somehow the estimation.

vvasco, 2019-04-12:
Projecting the 2D keypoints in 3D using a filtered depth map provides this output:
 
[image]

Feet appear quite stable even in 3D.

closed 2019-04-12 by pattacini

pattacini, 2019-04-14:
It may be beneficial to drop in here an archive containing the scripts used for this analysis.

vvasco, 2019-04-15:
Here it is:
[visualize_3d_foot.zip](https://github.com/robotology/assistive-rehab/files/3079010/visualize_3d_foot.zip)