robotology/assistive-rehab #184: Extend yarpOpenPose to include feet

vvasco, 2019-04-11:
Given #179, we want to extend `yarpOpenPose` to include feet keypoints.

vtikha, 2019-04-16:
I have just merged the new version of [yarpOpenPose](https://github.com/robotology/human-sensing/pull/16/commits) that includes the new changes of openpose. I will leave this issue under review for a couple of days for testing and then will close it. 

@vvasco @pattacini

pattacini, 2019-04-16:
Super cool 🚀

vtikha, 2019-04-17:
putting it back `in progress` as it occurred to me that it needs a quick modification 🤔

vtikha, 2019-04-17:
all fixed as of [this commit](https://github.com/robotology/human-sensing/commit/43064f481e84b41238270da060349d7dc2bc2315). The part list is now sent correctly through the output port

vtikha, 2019-04-17:
I am closing this for now.

closed 2019-04-17 by vtikha