robotology/assistive-rehab #192: Implement verbal interaction

vvasco, 2019-05-06:
Examples of potential conversation can be found [here](https://github.com/robotology/assistive-rehab-storage/raw/master/misc/videos/video_tug.mpg): 
- the robot has to explain the test;
- the robot has to reply to potential questions related to the test, such as "at which speed should I move?", "can I use the walking stick?", "how many time should I repeat the test?"; 
- speech can be triggered by the "hand-raised-up" option.

Given these considerations, we have to understand which tool to use and start to create a potential conversation for a meaningful verbal interaction.

cc @vtikha

vtikha, 2019-05-06:
:thought_balloon: :speech_balloon:

The best option then for this is to use the architecture developed for the IBM JL. 
Do you confirm that we will be using English? If yes, then:

- [ ] Use IBM speech to text - Backup can be google speech to text

- [ ] Use `IBM services` to branch out all the possible responses

- [ ] Speech can be trigger by `hand raise` or even `hey R1`

- [ ] Decide what mic to use, `handheld` or the `imbedded ones` on R1

pattacini, 2019-05-06:
Unfortunately, the first language we've got to use is **Italian** 🤔

vtikha, 2019-05-06:
we therefore need to use google services for speech to text.

vvasco, 2019-05-06:
Ok so we would have:
- **Google speech to text** for dealing with **italian**;
- **IBM services cloud** for all possible responses;
- speech triggered by **hand raise** or **"hey R1"**
- I would say **R1 imbedded mics** with some constraints (interaction within some distance and no noise), to avoid equipping users with microphones?

vtikha, 2019-05-06:
I completely agree @vvasco , but the mics are terrible on R1. In any case, the mics are plug and play therefore I can switch and test both easily.

vvasco, 2019-08-02:
We decided to use **Google services** for interpreting potential questions. As we don't plan to implement a conversation for now, but rather a simple _natural_ question and answer mechanism,  we can avoid IBM services and eventually resort to them later to include a more complex interaction.

The speech interpreter will manage questions (triggered by "hey robot" or hand raise), interpret them and send a corresponding keyword to `managerTUG` (port name: `/managerTUG/speech-interp:i`), which will formulate the answer accordingly and send it to `iSpeak`.
For now, `managerTUG` manages the following keywords:
- `speed`: "At which speed should I move?"
- `aid`: "Can I use a walking stick?"
- `repetition`: "How many repetitions should I do?"
- `feedback`: "How good am I in doing this?"
- `unclear`: the question is not understood;
- `unknown`: the question is not in the list of possible questions and the answer is unknown.

**Important**: we have to think of how to manage that the question generates an event, which can occur at any time and of which the modules involved must be aware.

cc @vtikha @pattacini

vtikha, 2019-10-17:
The following [PR](https://github.com/robotology/assistive-rehab/pull/225) deals with this by adding:

- **googleSpeech**: a module that receives sound from a microphone and retrieves the speech transcript from google speech cloud services
- **googleSpeechProcess**: a module that receives the speech transcript and analyses the phrase to retrieve its structure and meaning to interpret the questions. This uses google language cloud services. This then sends out the required keywords for use in the ```managetTUG```

vvasco, 2019-10-18:
Done in #225.

closed 2019-10-18 by vvasco