robotology/assistive-rehab #198: Deal with the palen occluding robot's arm

pattacini, 2019-05-13:
Through a YARP port we need to let the system know that the "occlusion" has been added to the experiment so that robot will stop giving feedback.

This feature will be available both in SAR and VA mode.

vvasco, 2019-05-17:
Done in `feat/virtual-robot`.

closed 2019-05-17 by vvasco

vvasco, 2019-05-21:
Hi @ddetommaso @PHC313, we have a question about the panel. How should we deal with it during an experiment with the real robot? Should we add it __during__ the exercise or you foresee two __different sessions__ (training without the panel + training with the panel?)

ddetommaso, 2019-05-21:
Hi @vvasco, 
based on our originally idea the panels should be present in a different session of the experiment, but I would ask @PCH313 and @CescoWillemse to take care of the design of the sudy in #203

vvasco, 2019-05-21:
Thanks @ddetommaso