robotology/assistive-rehab #234: Adjust skeleton poses due to filtered depth

pattacini, 2019-10-31:
Once we have the skeletons retrieved from the filtered depth, we need to compensate for the incorrect depth by applying a kind of unwarping in order to adjust the skeleton pose.

pattacini, 2019-11-04:
It came out that with the following parameters, the filtered depth gives us skeletons with the correct metrics:
```ini
[depth]
enable                      true
kernel-size                 10
iterations                  5
min-distance                0.5
max-distance                10.0
```
| |
|:---:|
| Skeletons with unfiltered depth |
| ![no-filtering][1] |
| Skeletons with filtered depth |
| ![filtering][2] |

Thus, @vvasco could you please try out the options above within TUG experiments and report here any findings?

[1]: https://user-images.githubusercontent.com/3738070/68116447-25723980-fefb-11e9-9d3c-64afafef4364.gif
[2]: https://user-images.githubusercontent.com/3738070/68116462-315dfb80-fefb-11e9-897d-dbfdce7ae290.gif

vvasco, 2019-11-11:
With the following parameters:

```
[depth]
enable                      true
kernel-size                 4
iterations                  3
min-distance                0.5
max-distance                10.0
```

I obtained the following result:

[image]

The ankles look reasonably close to the lines when I stand on them. I checked the metric I use for line crossing and it works out with these parameters.

pattacini, 2019-11-11:
Awesome! Let's go this way then ✨ 
Could you please put those parameters in `master` for the TUG?

vvasco, 2019-11-11:
Sure!

pattacini, 2019-11-11:
Soon afterward, please close this issue with a pointer to the commit hash.

vvasco, 2019-11-11:
Done here c86e3b69e054939a6fa9b00498ded711acf04952

closed 2019-11-11 by vvasco