robotology/assistive-rehab #238: Investigate why a skeleton can get lost during a TUG

vvasco, 2019-11-04:
We want to have a deep understanding of why a skeleton can be lost during a TUG. This occurs typically when the robot navigates to reach the finish line or during the exercise while tracking the skeleton. The second case might occur when the skeleton approaches the robot and only few keypoints remain in the FOV. If this is the reason, this might be avoided in a bigger space, increasing the distance from the skeleton that the `navController` tries to keep.

pattacini, 2019-11-08:
@vvasco do we have an estimate of how good is the 3D reconstruction of the skeleton when the robot is moving the head and/or controlling the base during TUG?

Asking since I think the 3D points can get − hopefully only a bit − "blurred" because of gazing/navigation.

vvasco, 2019-11-08:
I'll check this out during the experiments for the analysis of the loss of the skeleton. 
For now, what I noticed is that the step length extracted is reasonable, but the impact of the potential skeleton flickering on this specific metric might be minor (the signal is also filtered out).

vvasco, 2019-11-12:
I report here three scenarios while performing a TUG. 

|  Scenario | TUG |
| ------------- | ------------- |
| **(1) No skeleton loss**  | [image] |
| **(2) Skeleton lost during navigation:** <br> - stiff rotation of the base | [image] |
| **(3) Skeleton lost while walking:** <br> - few keypoints in the FOV | [image] |

### (1) No skeleton loss
This is an example of the skeleton we have during a TUG, when everything goes smoothly and navigation, attention and user's walking do not affect the skeleton tracking.

### (2) Skeleton lost during navigation
This is an example of skeleton loss while the robot navigates towards the finish line. 
The base rotates quite fast to reach the target with an angle within the angular tolerance and the gaze does not counteract for it. This causes the skeleton tracking to be lost.

### (3) Skeleton lost while walking
This is an example of skeleton loss while walking, during a TUG execution.
We can see that when approaching the robot, few keypoints stay within the FOV. Therefore the threshold `keys-recognition-percentage` is not reached and the `skeletonRetriever` creates a skeleton with a new id.   
The distance between robot and skeleton target is set to `1.5 m` due to reduced space, but in a bigger space a higher distance might help to keep more keypoints in the FOV. 

Regarding the "blurring" during navigation/gazing, the skeleton is quite stable when the robot navigates towards the user, it flickers when there is a stiff rotation of the base.

vvasco, 2019-11-12:
We can reduce the drag by reducing the `velocity-angular-saturation` on `navController` (see c86e3b69e054939a6fa9b00498ded711acf04952).

closed 2019-11-12 by vvasco