robotology/assistive-rehab #268: Add aruco boards to gazebo

vvasco, 2020-03-17:
We can rely on [this repo](https://github.com/joselusl/aruco_gazebo) that adds aruco markers into `gazebo` and add the [boards](https://github.com/robotology/assistive-rehab/tree/master/modules/lineDetector/app/conf) we use for the demo.

vvasco, 2020-03-17:
I added the start and finish lines in `gazebo` in this [fork](https://github.com/vvasco/aruco_gazebo). 
I can see the lines in `gazebo`, but when I run `lineDetector` the pose is completely wrong, as the estimation flickers in the rgb image (top left) and of course the final pose in the 3D world (top right) is wrong:

[image].

We can see there is a weird flickering effect on the texture of the aruco board in the rgb image, however not present in `gazebo`. I think this spoils the estimation. 
I'm going to investigate how to remove this effect.

vvasco, 2020-03-18:
[This commit](https://github.com/vvasco/aruco_gazebo/commit/b7e8f31ad9c08e4364cf1bd0767a4c3516db9b28) fixes the flickering and the light diffusion. 
The aruco board in the sdf is modeled as a box with a texture loaded from a jpg. The problem was that the depth of the box was too small, causing the flickering in the image. I also removed light diffusion on the object.
Now the estimated line looks correct (the command `go_to_line` successfully guides the robot to the line's origin):

[image]

closed 2020-03-18 by vvasco

pattacini, 2020-03-18:
That's awesome @vvasco!
Perhaps it would be worth contributing to the upstream repo by opening up a PR.

vvasco, 2020-03-19:
Done in #270.