robotology/assistive-rehab #269: Add a moving human model in gazebo

vvasco, 2020-03-17:
We have two options:
1. rely on @valegagge [repo](https://github.com/valegagge/humanModels);
2. add an actor to gazebo, as described [here](http://gazebosim.org/tutorials?tut=actor&cat=build_robot).

The main difference comes into play when we want to move the model: 
1. option 1 requires a joystick to move the model.
2. option 2 creates a simple trajectory which the model follows. This option also allows the model to stand up and sit down using predefined meshes.

We can start using on option 2 since the trajectory the human has to do is quite easy (straight, back and forth). Then we might think to combine the two options for more complex trajectories.

valegagge, 2020-03-17:
Hi @vvasco ,
I implemented [here](https://github.com/valegagge/my_appls/tree/master/GenericObjJoystickControl) a simple module that can move a generic object, also a human model, in gazebo using a joystick.

Maybe it can be useful for you. If you need it, we can think to move it from my personal repo to a common one.

vvasco, 2020-03-17:
Hi @valegagge thanks for pointing this out! 
For the moment I'm exploring option 2 because the trajectory is rather simple and it allows stand up and sit down actions. 
But I think it might be nice at some point to combine this option with your solution using the joystick to allow free movements. I have a joystick at home (not the xbox one required by `cer_linux.ini`), but I'm not sure if it's possible (and how) to configure `joystickCtrl` in order to use a different controller. Do you have any hint on this? 
Once set the joystick, then I might try your module to move the human model.

vvasco, 2020-03-24:
For now, I'm sticking to option 2. 
Following [this](http://gazebosim.org/tutorials?tut=actor&cat=build_robot) tutorial, it's possible to include into the scenario a human sitting, standing and moving:
 
[image] 

`yarpOpenPose` works pretty well (on my system, it runs at ~`5 fps`).

Now when we launch `gazebo` with `tug-scenario.world`, the human model immediately starts to move, with a trajectory and a speed defined in `tug-scenario.world`. We can delay the start by setting `delay_start` in the `.world`, but it would be good to be able to trigger the start, for example by sending a command to a `rpc` port as we do for the real TUG. 
It would be also nice to be able to attach objects to the human, like walking sticks or walkers (some models available [here](https://app.ignitionrobotics.org/openrobotics/fuel/collections/Hospital)).
For having more control on such parameters (i.e. triggering the start, setting the moving speed, attaching objects) we might develop a plugin.

pattacini, 2020-03-24:
Simply awesome 🥇

closed 2020-03-24 by vvasco