robotology/assistive-rehab #271: Integrate aruco fixes in upstream repository

vvasco, 2020-03-23:
We want to open a PR on the [upstream repo](https://github.com/joselusl/aruco_gazebo) to integrate the fixes we did to the aruco markers in `gazebo` (light changes and flickering, see #268).

vvasco, 2020-03-23:
Done [here](https://github.com/joselusl/aruco_gazebo/pull/2).

housso97, 2022-07-05:
thank you for this very useful rep, however i wanted to know exactly the IDs of the aruco markers and the size of those so that my computer vision approach can take these details into consideration.

vvasco, 2022-07-13:
Hi @housso97 thanks for the interest! You can find the information in the sdf of the aruco models: 
[size](https://github.com/robotology/assistive-rehab/blob/4cfa251b79edf30473e0d2346c4f882c4dc8a9e8/app/gazebo/tug/aruco_board_start/model.sdf#L10)
[model name](https://github.com/robotology/assistive-rehab/blob/4cfa251b79edf30473e0d2346c4f882c4dc8a9e8/app/gazebo/tug/aruco_board_start/model.sdf#L3)

Hope this helps!

closed 2022-12-03 by pattacini