robotology/assistive-rehab #275: Integrate Actor changes in upstream repo

vvasco, 2020-04-21:
We want to open a PR [here](https://github.com/osrf/gazebo) to include the changes we made to the `Actor` class (update sdf and play methods).

traversaro, 2020-04-22:
One thing that may come up (but you can probably discuss this directly with upstream developers) is that given that Gazebo 11 is the last major release of Gazebo, they do not accept in any branch changes that break the ABI of the public Gazebo functions (see https://community.kde.org/Policies/Binary_Compatibility_Issues_With_C%2B%2B#The_Do.27s_and_Don.27ts for a list of changes that affect the ABI). To avoid changing the ABI, you can: 
* Add new attributes not directly to the Actor class, but to the ActorPrivate class that implements the [PIMPL](https://en.cppreference.com/w/cpp/language/pimpl) pattern. 
* Instead of overloding an existing function, create a new function `PlayWithAnimationName` . You can  still avoid code duplication by calling  the new function `PlayWithAnimationName` in the implementation of the old function `Play`.

pattacini, 2020-04-22:
Thanks a lot @traversaro for the very useful heads-up.

vvasco, 2020-04-22:
Thanks a lot @traversaro, I'll update my changes as you suggested

closed 2022-12-03 by pattacini