robotology/assistive-rehab #292: Update YARP portmonitor initialization

pattacini, 2021-04-20:
Hi @vvasco 

The breaking changes introduced in https://github.com/robotology/yarp/pull/2552 will have an impact on our codebase.
For example, look at the following:

https://github.com/robotology/assistive-rehab/blob/d1b65f38390f34626155a457ed9caa8d11157bd8/app/scripts/AssistiveRehab.xml.template#L205

Would you be willing to take care of this update?

vvasco, 2021-04-21:
Hi @pattacini, I did the changes in the branch `fix/portmonitor-names`.

pattacini, 2021-04-21:
Fantastic! When the change will land in yarp@master we'll open up the internal PR.

randaz81, 2021-04-21:
Hi! After this reorganization, I am adding a couple of new portmonitors. Since you are already using zfp (**lossy** depth compression), I just wanted to tell you that a new pm, based on zlib, will be available, for **lossless** depth compression. The name will be `depthimage_compression_zlib`. Stay tuned ;-)

pattacini, 2021-04-21:
Thanks @randaz81 for letting us know 👍🏻 
I think we can already fix the pm strings after the merge of the PR in YARP while keeping an eye wide open on your current development.

pattacini, 2021-04-21:
This is the PR @randaz81 was referring to: https://github.com/robotology/yarp/pull/2553.

closed 2021-04-22 by pattacini