robotology/assistive-rehab #296: Dependencies for cornerRefinementMethod

gianscarpe, 2021-05-13:
Hi guys,
I'm trying to compile this repository. I installed all the dependencies but I still get the following issue:
```
/home/user1/assistive-rehab/modules/lineDetector/src/main.cpp: In member function 'virtual bool Detector::configure(yarp::os::ResourceFinder&)':
/home/user1/assistive-rehab/modules/lineDetector/src/main.cpp:160:26: error: 'struct cv::aruco::DetectorParameters' has no member named 'cornerRefinementMethod'; did you mean 'cornerRefinementWinSize'?
         detector_params->cornerRefinementMethod = cv::aruco::CORNER_REFINE_SUBPIX;
                          ^~~~~~~~~~~~~~~~~~~~~~
                          cornerRefinementWinSize
/home/user1/assistive-rehab/modules/lineDetector/src/main.cpp:160:62: error: 'CORNER_REFINE_SUBPIX' is not a member of 'cv::aruco'
         detector_params->cornerRefinementMethod = cv::aruco::CORNER_REFINE_SUBPIX;
                                                              ^~~~~~~~~~~~~~~~~~~~
/home/user1/assistive-rehab/modules/lineDetector/src/main.cpp: In member function 'bool Detector::detect_helper(const cv::Mat&)':
/home/user1/assistive-rehab/modules/lineDetector/src/main.cpp:676:118: error: too many arguments to function 'int cv::aruco::estimatePoseBoard(cv::InputArrayOfArrays, cv::InputArray, const cv::Ptr<cv::aruco::Board>&, cv::InputArray, cv::InputArray, cv::OutputArray, cv::OutputArray)'
 id=cv::aruco::estimatePoseBoard(corners,ids,curr_board,cam_intrinsic,cam_distortion,rvec,tvec,true);
                                                                                                   ^
In file included from /home/user1/assistive-rehab/modules/lineDetector/src/main.cpp:21:0:
/usr/include/opencv2/aruco.hpp:347:18: note: declared here
 CV_EXPORTS_W int estimatePoseBoard(InputArrayOfArrays corners, InputArray ids, const Ptr<Board> &board,
                  ^~~~~~~~~~~~~~~~~
modules/lineDetector/CMakeFiles/lineDetector.dir/build.make:83: recipe for target 'modules/lineDetector/CMakeFiles/lineDetector.dir/src/main.cpp.o' failed
make[2]: *** [modules/lineDetector/CMakeFiles/lineDetector.dir/src/main.cpp.o] Error 1
CMakeFiles/Makefile2:630: recipe for target 'modules/lineDetector/CMakeFiles/lineDetector.dir/all' failed
make[1]: *** [modules/lineDetector/CMakeFiles/lineDetector.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```
Can you please help me on this in using the right dependency for this function?
Thanks :)

gianscarpe, 2021-05-13:
Update:
I had a new issue related to the gazebo version, as during compilation I get
```
interface.cpp:138:16: error: 'class gazebo::physics::Actor' has no member named 'PlayWithAnimationName'; did you mean 'prevAnimationTime'?
         actor->PlayWithAnimationName(starting_animation);
```
Which version should I use?
I'm working with `Gazebo multi-robot simulator, version 11.5.1`

vvasco, 2021-05-14:
Hi @gianscarpe thanks for the interest in our repo!
At the moment, the latest `master` relies on [this](https://github.com/vvasco/gazebo) fork of `gazebo`, which includes some changes to the `Actor` class required to run the [TUG demo](https://robotology.github.io/assistive-rehab/doc/mkdocs/site/tug_demo/#running-the-tug-demo-on-gazebo) in `gazebo`. We will have to open a PR for integrating such changes (see #275).

Let me also point you at the website, which includes some [tutorials](https://robotology.github.io/assistive-rehab/doc/mkdocs/site/tutorial_intro/) on how to use / run modules or specific demos.

closed 2021-05-18 by gianscarpe