robotology/assistive-rehab #297: `lineDetector` and `skeletonRetriever` are to rely on I/F to get depth camera's params

pattacini, 2021-06-12:
As a result of https://github.com/robotology/yarp/pull/2596, `lineDetector` and `skeletonRetriver` don't build correctly (see https://github.com/robotology/assistive-rehab/runs/2807875607?check_suite_focus=true).

The problem is that now YARP no longer exposes the `VOCAB`'s required to retrieve params from the depth camera.
We would then need to access those values by resorting to I/F.

cc @vvasco @vtikha @drdanz

drdanz, 2021-06-15:
Sorry, I did not expect these vocabs to be used outside the wrapper and clients, @pattacini do you think we should partially revert these changes?

pattacini, 2021-06-15:
No problem @drdanz 👍🏻 
We've already a PR for fixing this natively: see #298.
Perhaps, you'd be willing to review it as I went on a bit open loop 😃