robotology/assistive-rehab #304: CI fails because of `curl` build error

pattacini, 2022-03-30:
Opened issue upstream: https://github.com/microsoft/vcpkg/issues/23874.

pattacini, 2022-03-30:
@traversaro gave https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/pull/51 another whirl. If successful, we will be able to update vcpkg.

pattacini, 2022-04-13:
The attempt came off well; we're now yielding a new release.

traversaro, 2022-04-13:
> The attempt came off well; we're now yielding a new release.

Release where the new archive will be uploaded: https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/releases/tag/v0.8.0 .
Job that is producing the archive: https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/actions/runs/2161357833 .

pattacini, 2022-04-14:
Everything works fine again as in https://github.com/robotology/assistive-rehab/actions/runs/2165904817.

Thanks heaps @traversaro!

closed 2022-04-14 by pattacini