robotology/assistive-rehab #306: Setup and test the TUG demo on R1SN003

vvasco, 2022-09-08:
This issue tracks the status of the tests of the TUG demo on the `R1SN003` robot.

cc @mfussi66 @pattacini @elandini84 @ste93 @lornat75

vvasco, 2022-09-08:
I report here the instructions to run the demo using docker.
The current development is on the branch `feat/docker`. Once merged, I will add the instructions on the official website.

To run the demo:
- start `yarp`
  - on **r1-base**:
    - `yarpserver`
    - `yarprun --server /r1-base`
  - on **r1-console**:
    - `yarprun --server /r1-console`
  - on **r1-face**:
    - `yarprun --server /r1-face`
  - on **r1-torso**:
    - `yarprun --server /r1-torso`

- run `yarprobotinterface` on **r1-base**:
  - `cd /usr/local/src/robot/robots-configuration/R1SN003`
  - `yarprobotinteface` 

- on **r1-console**:
  - `yarpmanager` 
    - open `AssistiveRehab_TUG` which includes the camera and iSpeak (the face is not powerful enough to use docker)
    - run all
  - cd tmp-valentina/etapas/assistive-rehab/docker/compose
    - docker-compose -f [docker-compose-console.yml](https://github.com/robotology/assistive-rehab/blob/feat/docker/docker/compose/docker-compose-console.yml) up
    - docker-compose -f [docker-compose-cuda.yml](https://github.com/robotology/assistive-rehab/blob/feat/docker/docker/compose/docker-compose-cuda.yml) up
    - docker-compose -f [docker-compose-speech.yml](https://github.com/robotology/assistive-rehab/blob/feat/docker/docker/compose/docker-compose-speech.yml) up
- on **r1-base**:
  - cd tmp-valentina/etapas/assistive-rehab/docker/compose 
    - docker-compose -f [docker-compose-nav.yml](https://github.com/robotology/assistive-rehab/blob/feat/docker/docker/compose/docker-compose-nav.yml) up
   
Note: the connections with the compose take some time. They can be checked in the yarpmanager, by refreshing the application.

Current issues:
- the wifi button has a high latency. The time between the moment you press and the moment you receive the request on the r1-console is very high
- the realsense has `424x240 px` resolution, while `cer_gaze-controller` uses `320x240 px`. We need a conf file for the controller relying on `424x240 px`.

vvasco, 2022-09-08:
Yesterday we tried to setup the demo on the robot with @mfussi66, @elandini84  and @ste93.

We had some **issues in detecting the `start-line`** using a camera **resolution of `424x240px`**.
The **estimation** in the **2D image was always wrong** (the line and the robot were placed in approximately the same position as the tests done in July, i.e. line entirely visible including the white part and robot neck pitch at ~ 22 degrees):
- some **markers** of the board were **not found** at all;
- even if found the final estimation was **wrong**, usually **inverting the y-axis with the z-axis**.

When using the **resolution of `640x480px`** the estimation was very **good** instead:

<p align="center">
[image]
</p>

With this resolution I had many issues in the past with `yarpOpenPose` being too slow, so I would avoid it. 

I tried again in the late afternoon with **`424x240px`** and the line estimation was good, using the **maximum neck pitch (~28 degrees) and placing the robot farther from the line than usual**. 

<p align="center">
[image]
</p>

Maybe this was due the combination of the camera / robot position and the changed light conditions? Maybe the color of the floor affects the detection? 
I don't get what's changed compared to the tests done in July though. 

I did some more tests using the standalone realsense (D415) at a resolution of `424x240px`, placing the **camera in different positions** and **changing the line position** to vary the light conditions.
The results were always good, here some examples:

| <p align="center"> [image] </p> |   <p align="center">  [image] </p> |  <p align="center">  [image] </p> |
| ------------- | ------------- |  ------------- |
| <p align="center"> [image] </p> |   <p align="center">  [image] </p> |  <p align="center">  [image] </p> |
| <p align="center"> [image] </p> |   <p align="center">  [image] </p> |  <p align="center">  [image] </p> |

We will do another round of tests on Monday morning on the robot and verify this again. Maybe we just need to move the robot neck further down than usual.

pattacini, 2022-09-08:
I've created the label https://github.com/robotology/assistive-rehab/labels/%F0%9F%8E%93%20etapas for the tasks aimed to the project.

lornat75, 2022-09-08:
about using 640x480 and interaction with opepose, maybe you can downside the images again when feeding them to openpose?

vvasco, 2022-09-08:
> about using 640x480 and interaction with opepose, maybe you can downside the images again when feeding them to openpose?

I think this could be a valid option, @ste93 also proposed the same.
I would give a second chance to the lower resolution anyway and if this does not work go with this alternative.

vvasco, 2022-09-15:
With @mfussi66 @ste93 @elandini84 we ran few tests in the last days.
This is the list of issues we encountered and a description of how we solved / bypassed:
- the **camera** mounted on the top of the robot head **could not load all the resolutions**, in particular `424x240` for depth, which can be listed using [`rs-enumerate-devices`](https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices). We changed the port it was connected to, at some point it started working (even using the same port that was not working before)
- **`424x240` depth images had artifacts** (horizontal bands) which made them unusable (`640x480` were fine instead). Swtching off the lights did not help. We started using the **realsense mounted in the robot** head which supports **`320x240`** resolution and does not have artifacts
- sometimes we still had **issues** when detecting the **lines**. Relaunching `lineDetector` resulted in a good estimate without changing the relative position line / robot
- the **3D skeletons** were **flickering**, probably due both to bandwidth saturation and high CPU load ( the majority of the modules were running on the robot console). Saving bandwidth and **balancing the CPU load** (running `docker-compose-console` on the second PC), solved the issue. We **saved bandwidth** as follows:     
    - applying [depth quantization](https://github.com/robotology/assistive-rehab/blob/fix/etapas-demo/app/conf/realsense2.ini#L18) and [compression with zlib](https://github.com/robotology/assistive-rehab/blob/feat/docker/docker/compose/docker-compose-console.yml#L129)
      note: in the past we used `zfp`, which gave use [some issues](https://github.com/robotology/assistive-rehab/issues/211)
    - in `skeletonRetriever` and `lineDetector` we open the rgbd device to retrieve the camera parameters. However we noticed that it keeps receving depth images from the camera, which we don't use (we get them from `yarpOpenPose`). [Closing the rgbd device](https://github.com/robotology/assistive-rehab/commit/e6f7d5f25c24367fad69ed92dbfbe13187d2e53b) after the camera parameters are retrieved solves this issue
- sometimes the **joystick does not start properly**. Unplugging and replugging the dongle on the base fixes this
- as per [@mfussi66 comment](https://github.com/robotology/assistive-rehab/issues/306#issuecomment-1246971409), we can use the **wifi button** with an acceptable latency only if the **robot is connected to internet**. We connected it through ethernet cable. 

We finally managed to run several times the demo successfully. This is a video showing the functionality (very bad quality but it's just to show the current status):

https://user-images.githubusercontent.com/9716288/190361504-ad5eb2d3-b0ce-4b89-8dd1-85b805d2dd43.mp4

Note: @mfussi66 raised the hand from the beginning to have the robot focusing on him rather than on iCub3 (which is also detected as a skeleton)

I'll open the PRs for [`fix/etapas-demo`](https://github.com/robotology/assistive-rehab/tree/fix/etapas-demo) and [`feat/docker`](https://github.com/robotology/assistive-rehab/tree/feat/docker) and update the instructions on the website.

Open points:
-  when the neck's pitch goes above ~ 25 degrees, the neck goes down and does not go back up when commanded (e.g. through motorgui). We manually bring it back up. There is already a [ticket open](https://github.com/robotology/icub-tech-support/issues/1426)
- we need an external microphone for dealing with user questions. In the video for testing purposes @mfussi66 was executing the TUG and I was talking to the microphone running on the robot console
- connecting the robot to internet through ethernet cable is not the best when the robot navigates. @ste93 told us that they used to have a dedicated mobile phone connected to IIT wifi and the robot connected in tethering through usb-c. The phone was placed on the robot through a phone armband

pattacini, 2022-09-15:
Nice progress 👍🏻 

A few comments below.

>sometimes we still had issues when detecting the lines. Relaunching lineDetector resulted in a good estimate without changing the relative position line / robot

I think this still represents a problem to solve. If it depends on the resolution of the camera somehow (to be clarified) then we should definitely consider increasing the resolution and/or decide which camera to pick (remember that we need to account for the correct transformation if we go with the top-head camera). In this respect, this further [comment](https://github.com/robotology/assistive-rehab/issues/306#issuecomment-1240790527) holds.

Also, the camera goes along with the quality of the data we can estimate that is relevant to the experiment. I believe this will be tested in the upcoming days.

>as per https://github.com/robotology/assistive-rehab/issues/306#issuecomment-1246971409, we can use the wifi button with an acceptable latency only if the robot is connected to internet. We connected it through ethernet cable.

I was told that the latency may still reach 3 seconds in some conditions, which is kind of slow.

>connecting the robot to internet through ethernet cable is not the best when the robot navigates. @ste93 told us that they used to have a dedicated mobile phone connected to IIT wifi and the robot connected in tethering through usb-c. The phone was placed on the robot through a phone armband

This could be a good alternative indeed.

pattacini, 2022-09-15:
We also need to address the problem with the wheels' grip material: https://github.com/icub-tech-iit/tickets/issues/2355.

mfussi66, 2022-09-15:
> We also need to address the problem with the wheels' grip material: [icub-tech-iit/tickets#2355](https://github.com/icub-tech-iit/tickets/issues/2355) (⚠️ private link).

We arrived yesterday in the robot arena and the grip was glued back on, it seemed quite sturdy and reliable.

pattacini, 2022-09-15:
>We arrived yesterday in the robot arena and the grip was glued back on (not sure who did it), it seemed quite sturdy and reliable.

Maybe @fbiggi?

Speaking with @maggia80 it came out that we could think of redoing the wheels with a proper gripping footprint. Don't know how long would it take though.

vvasco, 2022-09-15:
> I think this still represents a problem to solve. If it depends on the resolution of the camera somehow (to be clarified) then we should definitely consider increasing the resolution and/or decide [...]

I agree this has to be fixed. This happened with both cameras (the one on the top of the head and the one in the robot's head) at low resolution. It's very weird to me that `lineDetector` was not able to estimate the line and after relaunching the module the estimate was good, given that the position of the robot and the line did not change and the light conditions were the same. Probably we can go for [this option](https://github.com/robotology/assistive-rehab/issues/306#issuecomment-1240790527) then.

> I was told that the latency may still reach 3 seconds in some conditions, which is kind of slow.
> 

Yes sometimes this also happened, and it used to happen also in the past. I think the latency gets worse when the battery charge starts decreasing. Before running the tests we could make sure that the button is completely charged.

pattacini, 2022-09-15:
>t's very weird to me that lineDetector was not able to estimate the line and after relaunching the module the estimate was good, given that the position of the robot and the line did not change and the light conditions were the same.

Some "memory effect"? Meaning that throughout operations we run code branches that we don't visit at startup, modifying somehow the logic of the check? If so, it's not even said that increasing the resolution will be resolutive. Just speculating.

>Yes sometimes this also happened, and it used to happen also in the past.

👍🏻 

>I think the latency gets worse when the battery charge starts decreasing. Before running the tests we could make sure that the button is completely charged.

Ok, so we could monitor this behavior by trying to do the experiments always with a good charge level.
We can also buy a second button, just in case. What do you think?

randaz81, 2022-09-15:
`they used to have a dedicated mobile phone connected to IIT wifi and the robot connected in tethering through usb-c. The phone was placed on the robot through a phone armband`
It's possible, of course, but do you have issues using R1's standard router on a desk? Here is the connection diagram:

 ```
ROBOT ---- wifi ------ router ------ internet
                          |---------- laptop1
                          |---------- laptop2
                         etc....
```

vvasco, 2022-09-16:
> > t's very weird to me that lineDetector was not able to estimate the line and after relaunching the module the estimate was good, given that the position of the robot and the line did not change and the light conditions were the same.
> 
> Some "memory effect"? Meaning that throughout operations we run code branches that we don't visit at startup, modifying somehow the logic of the check, slightly but decisively sometimes? If so, it's not even said that increasing the resolution will be resolutive. Just speculating.

I need to investigate on this. So far at higher resolution I've never seen the same issue, but it's also true that I ran less tests.

> 👍🏻
>
>
> Ok, so we could monitor this behavior by trying to do the experiments always with a good charge level. We can also buy a second button, just in case. What do you think?

Definitely. I have another mystrom button (I bought it long time ago for testing). We can configure it on the robot wifi and alternate it with the other.

> `they used to have a dedicated mobile phone connected to IIT wifi and the robot connected in tethering through usb-c. The phone was placed on the robot through a phone armband` It's possible, of course, but do you have issues using R1's standard router on a desk? Here is the connection diagram:
> 
> ```
> ROBOT ---- wifi ------ router ------ internet
>                          |---------- laptop1 (no wifi)
>                          |---------- laptop2 (no wifi)
>                         etc....
> ```

thanks @randaz81, that would be another possibility

pattacini, 2022-09-16:
>I have another mystrom button (I bought it long time ago for testing). We can configure it on the robot wifi and alternate it with the other.

Cool! Let's do it 👍🏻

lornat75, 2022-09-20:
about the microphones as you probably know on the 5gtours project we used an external mic mounted on the head of R1, it worked ok in our experiments, provided people spoke with the correct timing and within the microphone "field of view". We have been also testing other OEM microphones for better integration but I think this may still need some time.

lornat75, 2022-09-20:
nice progress though!

mfussi66, 2022-09-28:
Today in the morning we tackled the motionAnalyzer: we started off by tweaking the setup of the Yarp modules to save an stabilize the bandwidith as much as possible, to make sure that the skeleton data would be reliably received.
However, the step length analysis is still not reliable. We will check if changing the skeletonRetriever parameters can help, but we are having doubts about the reliability of the step length computation.

pattacini, 2022-09-28:
>However, the step length analysis is still not reliable. We will check if changing the skeletonRetriever parameters can help, but we are having doubts about the reliability of the step length computation.

A couple of related resources:
- #178 
- #279

pattacini, 2022-09-28:
It may be convenient to record a data stream to recreate the skeleton on the same data offline and analyze what's happening in detail, with approaches similar to those in #178 and #279.

The effect of tuning of the `skeletonRetriever`'s parameters could be judged offline too.

pattacini, 2022-09-28:
These params are for simulation, don't we have analogous ones for the real TUG experiments that we conducted in the kitchen?

Be careful as `filter-keypoint-order` is `int32` so we better ought to specify `--filter-keypoint-order 5` to avoid incurring into the "conversion hell".

vvasco, 2022-09-28:
I mentioned the parameters used in simulation because we benchmarked the metrics on that scenario. For the real TUG experiments in the kitchen I used [these values](https://github.com/robotology/assistive-rehab/blob/f37f5c8284e0198ab926750a1187083957c751fa/app/scripts/AssistiveRehab-TUG.xml.template#L56), but it was with a different camera, we probably need some fine-tuning.

mfussi66, 2022-09-30:
Here is a dataset dump that can be used to replay the yarpopenpose and depth data, so offline processing can be performed:

https://istitutoitalianotecnologia-my.sharepoint.com/:u:/g/personal/mattia_fussi_iit_it/EWW_5qRAvHlMs0IozvHZphAB7-tgfKnWDKcQjdSq5dpuMg?e=KSashd

I created a branch in which I added a docker-compose-replay, that launches only the components for post-processing, mainly yarpDataPlayer and motionAnalyzer. 

Instructions for posterity on how to use docker-compose-replay:
1. open the application AssistiveRehab-replay on the Yarpmanager 
2. launch the compose with `docker-compose -f docker-compose-replay.yml up` -> the compose will launch a container with a yarprun --server /replayer
3. on the YarpManager launch `skeletonRetriever` 
4. load the dataset on the yarpDataplayer
5. make all the connections in the YarpManager
6. Press play on the Yarpdataplayer
7. open a terminal with `yarp rpc /motionAnalyzer/cmd` and launch the commands
     1.   loadExercise tug
     2. selectMetric step_0
     3. selectMetricProp step_length
     4. selectSkel <number you see on the skeletonviewer>
     5. start false
     6. then stop in the end

pattacini, 2022-09-30:
Nice work @mfussi66 👍🏻 
My only point is that it'd be much better to post the dataset on https://github.com/robotology/assistive-rehab-storage.

mfussi66, 2022-10-10:
To recap the past days: 

- `motionAnalyzer` is now able to work offline, the relevant PR was merged: https://github.com/robotology/assistive-rehab/pull/318
- @elandini84 successfully modified the `navController` to make it work offline: https://github.com/robotology/assistive-rehab/commit/9b8dd3f263bae206346b97dd06eab03028bf1089 It was tested successfully offline, we will open a PR when we will run a TUG demo.

1. On Friday, I inspected the `skeletonRetriever` skeleton data structure that is outputted towards the `skeletonViewer`, and noticed that the computed ankles distance is compatible with the expected value, while taking a known step. E.g., if the step I took was 50cm, the `skeletonRetriever` would return ankle poses that would result in a ~44cm step. 
Besides the 10% discrepancy, the results were consistent when taking other steps. 

2. Therefore, today me and @ste93  took a look at the `motionAnalyzer`. We did not really want to touch the code, but we could not really understand why some specific operations of the step length estimation were performed. 
So @ste93 simplified the approach and added a little bit of filtering to the `FindPeaks` function (actually there is already a prefilter that we could retune), and now the step length waveform follows the person much better 👍 I'll show a video of it ASAP.

However, what needs to be ironed out now is the error wrt the ground truth increasing as the step length increases. This happens in the `skeletonRetriever` too. But playing around with the parameters might be a solution.

mfussi66, 2022-10-19:
An update regarding the past days:

Although very simple scenarios considered successful (single steps, perfectly placed in front of the robot), the evaluated metric gets less reliable when the person walks normally, by putting one foot in front of the other.
We did some tests and I extracted the positions of the ankles.
the test was the following:
1. 30cm steps moving from right to left (wrt the robot)
2. 30cm steps moving from left to right
3. repeat the two above with 60cm steps

This the resulting skeleton, as seen by the skeletonViewer. The grid SHOULD be spaced 0.1x0.1m, but I need to verify that in a static scenario.
The interesting part of the video starts at 00:40

https://user-images.githubusercontent.com/38140169/196751918-b647f7d6-6fd0-42e7-bba4-f9acc838842c.mp4

By reading the position of the ankles, I get the following plots.

[image]

Some comments:
- the distance covered by each ankle should be double the step size, but it is not
- Even though the amplitude of the ankles' position should be double of what is expected,  we can clearly see the steps being made, so there is some useful data there that can be extracted
- the difference between the two ankles along the y axis (bottom left plot) is promising, because the amplitude of the waveform kind of matches the steps
- We need to be really careful when gathering data, changing the parameters of the skeleton retriever for depth processing does not seem like a silver bullet

mfussi66, 2022-10-21:
> This the resulting skeleton, as seen by the skeletonViewer. The grid SHOULD be spaced 0.1x0.1m, but I need to verify that in a static scenario.

I just verified it  as a sanity check: what we see in the SkeletonViewer is subjects in the space expressed in meters. If i place myself statically 3m from the robot, i can see in the virtual world that I am 3squares distant from the robot. Therefore the grid it's 1mx1m, and we can use a smaller resolution, like 0.1mx0.1m to further check the step distances in the skeletonViewer.

pattacini, 2022-10-21:
Since there might be a nonlinear mapping from the realsense and the real world, it could help to do this sanity check also at different increasing distances.

If there's a need for a look-up table to compensate for the nonlinearities, @randaz81 suggests that we could use the LIDAR to do the remap automatically online.

mfussi66, 2022-11-07:
Extensive tests in the past week highlighted the necessity to use the Realsense D450 mounted on top of the robot head, since the one inside has a range of only 3meters. The main culprit of the inexact metrics was indeed the lower-range camera.

Additionally, it was necessary to project the evaluated metric (for example the step length) on the skeleton planes: sagittal, coronal, and transverse. This is now possible, although we need to be careful about the orientation of the person wrt the robot.
It seems that the more the person is in front of the robot, the more the metrics are reliable. Which should be consistent with openpose, since a person standing up on the side would have half of the body occluded wrt the camera.

See the video below. In the scope red is the step length and green is the step width:

https://user-images.githubusercontent.com/38140169/200041417-85a52a56-ce42-4ede-9c45-23e426abe932.mp4

As you can see when walking towards the robot, the metrics are accurate.

I think for the TUG demo we could keep the scenario of the person walking towards the robot, so that the whole pipeline tests can resume.

mfussi66, 2022-11-25:
Latest trials revealed that it could be more robust to measure the step distance of the subject as the norm of the cartesian distance between the ankles. In this way, we could mitigate the metric's sensitivity wrt the subject's orientation. 
See the video here:

https://user-images.githubusercontent.com/38140169/204008489-c919c66a-26e6-4cb2-87b9-8d51cc3e6d70.mp4

And the related PR: https://github.com/robotology/assistive-rehab/pull/323

Next, I will see if it's possible to change the metric by sending commands to the motion analyzer with `selectMetricProp`.

mfussi66, 2022-12-05:
On thursday we launched the demo successfully and without after implementing the many fixes in the PR below:

https://user-images.githubusercontent.com/38140169/205602167-4b4269af-621f-450e-b725-b9db5c0b009e.mp4

https://github.com/robotology/assistive-rehab/pull/328

Though we will check again if the the event collection properly records all the agreed events.

pattacini, 2022-12-05:
Nice!

It seems though that when @ste93 turned around to reach back the seat the system got paused somehow.

Not sure about the following 2 points:
1. Is the graph on the left showing the number of steps correctly?
2. It looks to me that the video on the right (bottom) is definitely off-sync with the real stream.

Did you notice that too? Is everything under control?

lornat75, 2022-12-05:
looks great! I also had issues understanding the plot...

mfussi66, 2022-12-05:
> Nice!
> 
> It seems though that when @ste93 turned around to reach back the seat the system got paused somehow.
> 
> Also, not sure about the following points:
> 
>     1. Is the graph on the left showing the number of steps correctly?

It shows the step length which might need a little more filtering for better highlighting of the step profiles.

>     2. It looks to me that the video on the right (bottom) is definitely off-sync with the real stream.

Yes the video in the yarpview is very laggy, though the measured framerate is stable at 10FPS. We will investigate it. It might be caused by  OpenPose that takes most of the video memory.

mfussi66, 2022-12-12:
Today we ran the demo again, and successfully uploaded the json file to the online platform:

[image]

The column "Event name" is hardcoded in the python script, we shall change it to something more generic and user-specific.

A little bugfix PR was opened after the tests: https://github.com/robotology/assistive-rehab/pull/330

> 2. It looks to me that the video on the right (bottom) is definitely off-sync with the real stream.

With @elandini84 I noticed that it was caused by dragging the window from one screen to the other. Most likely an Ubuntu or Qt issue with high refresh rate screens, definitely not yarp.

mfussi66, 2022-12-16:
This video is another demo in which we reduced the distance between robot and subject to 1.8mt, to handle the small space of the arena. It should be clearer, and we could show something like this to the next monthly meeting.

https://user-images.githubusercontent.com/38140169/208148026-49e5b011-fff8-42d7-a6de-43b3b21fa8f3.mp4

pattacini, 2022-12-16:
Nice! The demo runs much smoothly 🎉 

The steps detection performs better when @elandini84 goes back to the chair. We can clearly see the steps and the gait is fluid. Instead, when he reaches the finish line in the first part, his gait turns out to be a bit wobbling because R1 is moving backward slowly. As a result, the recognized steps do not look as nice as in the second stint.

Probably, this can be solved by tuning the human-robot distance slightly more and/or increasing the robot speed.

pattacini, 2023-01-25:
Better off closing this longstanding issue now that we have a quite solid bunch of components to go in favor of narrower tasks.

cc @mfussi66

closed 2023-01-25 by pattacini