robotology/assistive-rehab #307: Develop the connection Navigation ➡️ GRACE

pattacini, 2022-09-08:
In the context of the ETAPAS project, we're required to provide navigation data to the GRACE platform, pretty much as done for the speech.

mfussi66, 2022-09-21:
Today, me and @vvasco started off the task. We successfully implemented the logic in the `eventCollector` module regarding the collection of `obstacleDetector` info. The `eventCollector` is tasked to populate a JSON file with info regarding errors from the speech and navigation pipeline.

So far, an error event is written in the JSON file every time the `obstacleDetector` communicates the presence of an obstacle inside a radius. The radius is configurable, so we can tune it to detect only potential contact against the environment.
We just need to make sure that the eventCollector does not flood the file with lines. This can be solved, for example, by making the obstacleDetector send only the obstacle found/removed events, instead of a stream of messages.

I think we just need to clarify if there are other faulty events regarding navigation that can be useful to GRACE.

mfussi66, 2022-09-22:
Today we checked the behaviour of the `eventCollector` in presence of both `obstacleDetector` warnings and `googleSpeech` warnings on the robot. The tests went well, there is just an issue regarding the way the `jsoncpp` library is used, that necessitated a suboptimal workaround to make everything work as expected. It might need a little bit of study work in a flexible way as desired.

closed 2022-09-24 by pattacini