robotology/assistive-rehab #309: Host docker images on GitHub/robotology

pattacini, 2022-09-12:
Hi @vvasco 

I can take care of producing the necessary GH actions to build and host the docker images we need at robotology.

Also, let's keep for the moment the dependencies from `icubteamcode` images hosted on Docker Hub. We will streamline them later.

Can you only please tell me which images we're required to build among those available at https://github.com/robotology/assistive-rehab/tree/feat/docker/docker?

pattacini, 2022-09-14:
@vvasco, could you also please share here the signature of the command line you've been using to build these images (i.e., build args, tags...).

vvasco, 2022-09-14:
> @vvasco, could you also please share here the signature of the command line you've been using to build these images (i.e., build args, tags...).

I just set the images tag with no additional options, therefore:
- `docker build -t icubteamcode/fdg-basic:v2022.05.2_sources .`
- `docker build -t icubteamcode/fdg-navigation:v2022.05.2_sources .`
- `docker build -t icubteamcode/fdg-speech:v2022.05.2_sources .`
- `docker build -t icubteamcode/fdg-gazebo:v2022.05.2_sources .`