robotology/assistive-rehab #331: Investigate the possibility to replace the Wi-Fi buttons

mfussi66, 2022-12-21:
The [MyStrom Wifi Button](https://mystrom.com/it/wifi-button/) that is now used to trigger the speech interaction in the TUG demo takes a long time to configure, needs a phone app, and requires internet connection to work in a reasonable timeframe. Additionally, it requires dedicated yarp JS code.

It might be useful to investigate if it can be replaced by a simpler device, for example a bluetooth joystick. Joysticks can be easily handled by yarp, since the dedicated device is already written, and the configuration supports launching rpcs natively (through shell scripts). 

cc @pattacini @randaz81 @elandini84 @ste93 .

mfussi66, 2023-01-10:
This little device called 8BitDo works well as a bluetooth joystick on Linux, we can configure it to send speech triggers through one of the buttons (like the right top one)

[image]

mfussi66, 2023-01-17:
Just created the proper config file that mimicks the wifi button, will open a PR soon:

https://user-images.githubusercontent.com/38140169/212932268-3982e0e2-f4c9-455f-9c2c-85611a4e4e1c.mp4

pattacini, 2023-01-18:
Integrated via:
- #343