robotology/assistive-rehab #340: Improve the peak finding algorithm to improve gait estimation

mfussi66, 2023-01-10:
The peak finding algorithm is used by the `motionAnalyzer` module to detect in real time the number of steps that the patient does while walking. This information is also used to estimate cadence and speed of the subject.
The algorithm is also used in the reporting Python notebooks, for offline processing of the skeleton data.

The Python notebook uses a [Scipy function](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.find_peaks.html#scipy.signal.find_peaks), while the `motionAnalyzer` uses a [handcrafted one](https://github.com/robotology/assistive-rehab/blob/6e21c89a7e2cbe45654577b873f013ccab1eea19/modules/motionAnalyzer/src/Processor.cpp#L381).
Both implementations are quite sensitive to noise profile and configuration parameters (e.g. peak threshold).
To improve robustness of the procedure we could use better techniques, that can be tracked in this issue.

mfussi66, 2023-01-10:
# Continuous wavelet transforms

Peak-finding through CWT is a particular application of wavelet-based convolution. The wavelet is scaled in width multiple times and the local maxima for each scaled convolution are linked to each other. This is a reference paper: https://academic.oup.com/bioinformatics/article/22/17/2059/274284?login=false
It's implementation is included in Scipy and can be tested.

pattacini, 2023-01-13:
Another possibility, less advanced but easier to tune, is to resort to an SG filter as in https://www.mathworks.com/matlabcentral/answers/696995-find-peaks-valleys-of-a-noisy-signal.

mfussi66, 2023-01-16:
Scipy already provides the SG filter, so it was easy to implement. 

At the moment I used the following set of computations:
1. compute difference between ankles
2. apply SG filter to difference
3. compute norm2 to get distance between ankles
4. remove outliers outside 3 standard deviations using zscore, replace with 0.0
5. use the standard find_peaks function 

The parameters are:
- SG Filter:
    - window size 20 samples (2 seconds)
    - polynomial order 2 to have significant smoothing even though the window is quite small
- find peaks function
    - minimum height: 35cm
    - minimum horizontal distance  10 samples (1 second)

Here is a usage example with the small dataset used previously. The cliff at ~45sec is somewhat of an outlier, and the two peaks to the side should only by one.
[image]

Here is a bigger dataset with less forgiving data
[image]

The filter seems promising, however it needs to be verified if the peaks are actual steps. To do so, I will see if I can exhume a script posted in the previous issues that would reconstruct the skeleton from keypoints.

pattacini, 2023-01-17:
Really promising!
I reckon we can open a PR to integrate this latest tidbits.

mfussi66, 2023-01-17:
Pr here! https://github.com/robotology/assistive-rehab/pull/342

closed 2023-01-19 by pattacini